acfr/cam_lidar_calibration #1: catkin build failed: missing header file

hm2092, 2021-08-04:
Hello, First of all thank you for the ros package. When i try to build the package i get the error:
```bash
/home/user/catkin_ws2/src/cam_lidar_calibration/src/cam_lidar_panel.h:14:10: fatal error: cam_lidar_calibration/RunOptimiseAction.h: No such file or directory
```
Could you update the missing header file? Thank you in advance.

darrenjkt, 2021-09-04:
The file is there. Did you run `source devel/setup.bash` after `catkin_build`? I noticed that if you build and source a couple times, it eventually finds where the action header file is.

closed 2021-09-09 by darrenjkt

hm2092, 2021-09-29:
Hi @darrenjkt . Thank you for your reply. Sorry for the delayed update. You were right.

I got the error when i did `catkin build`. The error was occuring during the build process and crashes the build process. I have also done `source devel/setup.bash` before executing `catkin build`. I noticed that for some reason it was invoking `make -j1` and skipped building the RunOptimizeAction files. Invoking `catkin build -j4` solved my problem.