acfr/cam_lidar_calibration #3: Recording my own data

islamtalha01, 2021-09-01:
HI, 
          I want to know that how you recorded the bag file of your lidar stream and camera stream at the same time. I basically want to know that what are the steps to make your own data set for calibration.

darrenjkt, 2021-09-04:
You can record a rosbag using `rosbag record`. We simply specified the camera and lidar topics for recording as such: 
```
rosbag record -O calib_dataset /velodyne/front/points /gmsl/A0/image_color 
```

For more info, see this [tutorial](http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data).

islamtalha01, 2021-09-04:
I have done recording the data but I am facing an issue I am not able to
extract feature from the live fee of both sensors. The recorded file is
also not showing point clouds. my lidar is os1 64. i am trying to get
extrinsic parameter of calibration. kindly see my video on google drive so
you can understand the scenario.
https://drive.google.com/drive/folders/1lQMeMv0lzVR_JxMgSZebeir5DbGxhlOL?usp=sharing

closed 2021-09-21 by darrenjkt

islamtalha01, 2021-09-29:
Now i am able to extract the checkerboard but i am getting an error due to
which i am not able to calibrate. I have attached the error screenshot.
Kindly reply as soon as possible.