acfr/cam_lidar_calibration #4: Ouster lidar os1 64

islamtalha01, 2021-09-01:
Can this package be used for calibration of ouster os2 64 and Ip camera with this package? if it is possible then kindly guide me to complete the procedure I am doing as the research intern at NUST Pakistan. Kindly help me to achieve this task.

chinitaberrio, 2021-09-03:
Just checkin the [ROS driver for the Ouster lidar](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/include/ouster_ros/point.h), I think it should work as it has the ring field in the point time, which is a requirement for the usage of the package.

islamtalha01, 2021-09-03:
I am trying but if there is an issue I will let you know. kindly help me in
this regard.

islamtalha01, 2021-09-03:
I am trying with ouster os1 64 and USB camera. both lidar and USB camera
drivers are installed and they are also running pretty well individually.
I think I am mistaking in configuring my config file. I am attaching the
screenshot of the config file and please tell me where I am mistaking. I
also attached the warning I am getting and also my viz is not showing the
output of the camera and lidar when running the lidar_camera
calibration tool.

chinitaberrio, 2021-09-03:
I'm unable to see the attached files

islamtalha01, 2021-09-03:
kindly check the pics in the folder uploaded on google drive.
https://drive.google.com/drive/folders/1lQMeMv0lzVR_JxMgSZebeir5DbGxhlOL?usp=sharing

chinitaberrio, 2021-09-03:
You need to change the global frame in rviz to your lidar frame

islamtalha01, 2021-09-03:
There is no option to change the frame to ouster in that rviz window.
I think my topics in the cfg file are not correct.
How to find the exact topic of my sensor in ros. Kindly help me in this
regard.

chinitaberrio, 2021-09-03:
You should be able to change the global frame in rviz, refer to this document: http://wiki.ros.org/rviz/UserGuide
Once you have launched all your drivers, open a new terminal and type `rostopic list`, you should be able to see all the topics published by the drivers, you can also take a look at these links: http://wiki.ros.org/rostopic http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics

islamtalha01, 2021-09-03:
Thanks, a lot Stephany.
 I will look into it and come back to you. I am very thankful to you for
this help. I am newby that's why I am asking all these questions. again
thanks a lot.

islamtalha01, 2021-09-05:
> I am able to get the camera and lidar stream in the rviz window. now when
> I trying to extract features from a live feed. it is not happening. I have
> recorded the video kindly see the video and help me in this regard. i will
> be very thankful to you for this.
> please see video in the link below.
>
> https://drive.google.com/drive/folders/1lQMeMv0lzVR_JxMgSZebeir5DbGxhlOL?usp=sharing

Hello Stephany,
                          Is there any other way to communicate with you in
this regard? Facebook, WhatsApp, or MS teams. I am asking this because
there we can contact easily and often. I am a beginner, I just need your
help for some time otherwise there is no support of ROS with ouster in this
regard on the internet. kindly help me. I will not create inconvenience for
you.
Thank you.

chinitaberrio, 2021-09-05:
If you have any issues with your ouster driver please address them at: https://github.com/ouster-lidar/ouster_example, we can only support problems related to the camera-lidar calibration package.

islamtalha01, 2021-09-05:
I have no issue regarding driver i am just saying that theyvhave no support
regarding camera lidar calibration that's why I am asking about your
package.
Did you saw my video that i upload on the Google drive.

chinitaberrio, 2021-09-05:
Prepare a rectangular chessboard printout, preferably with odd-numbered grid dimensions (e.g. 9x7, 11x13)
Firmly attach the chessboard on a rigid, opaque, and rectangular board such that both their centres align and their edges remain parallel to one another.
**Choose a suitable stand that can mount the target with little to no protruding elements from the board's edges.**
Mount the target on the stand such that it is tilted at an angle of 45-60 degrees with respect to the ground plane.

You also need to adjust the experimental region for your case. The in rqt_reconfigure, feature_extraction variables correspond to the bounds of the 3D experimental region along the LiDAR's x,y,z axis:
x_min [rear], x_max [front] with range of [-6.0, 6.0]
y_min [right], y_max [left] with range of [-5.0, 5.0]
z_min [bottom], z_max [top] with range of [-2.0, 2.0]

islamtalha01, 2021-09-06:
What does it mesns by adjust the experimental region? How we can do it? We
have to edit some file?

darrenjkt, 2021-09-06:
When you run `roslaunch cam_lidar_calibration run_optimiser.launch`, two windows will pop up. One is Rviz, the other is the rqt_reconfigure window. The rqt_reconfigure window is where you can adjust the experimental region. That window basically sets the maximum and minimum x,y and z value such that you only allow certain points through within those bounds and filter the rest out.

islamtalha01, 2021-09-06:
Okkay got it. But the thing is that when i am changing this values in my
previous exp nothing happens. No change in region happens. What would be
the reason? I recorded bag file containing img and point cloud but when i
am running this file and adjusting my exp region nothing happens.
There is a error of no tranform found or something like what is this issue?

chinitaberrio, 2021-09-06:
Make sure the topic name is correct. Also you can refer to this readme file: https://github.com/chinitaberrio/cam_lidar_calibration

darrenjkt, 2021-09-06:
Before adjusting anything in the rqt_reconfigure, you should be able to see the entire pointcloud if you've configured RViz correctly

islamtalha01, 2021-09-06:
Darren
            I am seeing both lidar point clouds and camera stream by
playing my rec bag file.
But it is showing some issues of transform.
My bag file has no \tf topics.
Darren can we talk in something that is much fluent or easy like Skype or
ms teams.
I am doing an intern at university, In a hour i will have the sensor os1 in
my hand, can you help me with that?. On email it is difficult to mention
things in robust manner. It will be the matter of 10 to 20 minutes. But by
this way i will be sure in these minutes that this package is gonna work
for or not.
Kindly think about it and let me know, otherwise email will the option i
opt for. Kindly do me favor of 10 to 20 minutes.

islamtalha01, 2021-09-06:
I am able to get the camera and lidar stream in the rviz window. now when I
trying to extract features from a live feed. it is not happening. I have
recorded the video kindly see the video and help me in this regard. i will
be very thankful to you for this.
please see video in the link below.
https://drive.google.com/drive/folders/1lQMeMv0lzVR_JxMgSZebeir5DbGxhlOL?usp=sharing

chinitaberrio, 2021-09-06:
The first step is to have the proper setup:

> Prepare a rectangular chessboard printout, preferably with odd-numbered grid dimensions (e.g. 9x7, 11x13)
> Firmly attach the chessboard on a rigid, opaque, and rectangular board such that both their [...]

The tool provided won't work if you are holding the checkerboard as it needs to be isolated from the background by adjusting the experimental region like this: 

[image]

darrenjkt, 2021-09-06:
You can send me an email at jtsa5115@uni.sydney.edu.au and we can chat further.

islamtalha01, 2021-09-08:
> Is the problem of adjusting the rqt_reconfigure window but the point cloud
> range unchanged? I also encountered the same problem. Can you tell me how
> to solve it, thank you.

I am still facing this problem so if there any solution will come to me I
will tell you

chinitaberrio, 2021-09-08:
Can you show me the point cloud topics that appear on rviz? 
[image]

darrenjkt, 2021-09-08:
I've made some annotations for the rviz setup process. Let me know if this helps.

[image]

islamtalha01, 2021-09-08:
Thanks, I'll check it out.

closed 2021-09-21 by darrenjkt

islamtalha01, 2021-09-29:
Hi, Darren,
                  I have tried the things you said in the last reply but i
am not able to extract the checkerboard. some errors appeared which I
attached a screenshot. the second thing using recorded bag file How to
apply feature extraction? do i have to pause the bag file at some point? or
just use feature extraction when the rosbag file is playing?.

islamtalha01, 2021-09-29:
Now i am able to extract the checkerboard but i am getting an error due to
which i am not able to calibrate. I have attached the error screenshot.
Kindly reply as soon as possible.

jvnsg, 2022-06-16:
Hey @islamtalha01,
I have some similar issues like you had configuring the calibration tool for the Ouster LIDAR. Is there a way I can contact you directly to help me out?

KyunaCho, 2022-09-20:
> 

I'm using the os1 64ch lidar, too. but I'm not able to extract the chekerboard. How did you do?? Help me, please. Just like your situation before, I adjust the rqt_reconfigure window but the point cloud range unchanged.

testgithubme, 2022-11-14:
Hey @islamtalha01 ,
I am facing the problem in Ouster LIDAR and Camera calibration. can you help me solve the problem.....

mumu-o, 2023-10-08:
> did anybody get the solution of this issue? if so please help me to calibrate OS2 64 channel Lidar and camera .....
Hi @TunnaBaitha 
I got the same issue when using OS2 64 channel LiDAR, but I still didn't find a solution yet. Have you solved it?

chinitaberrio, 2023-10-08:
@mumu-o, What specific problem are you experiencing?

mumu-o, 2023-10-08:
> @mumu-o, What specific problem are you experiencing?

Hi @chinitaberrio,
Thank you for your reply.
I currently use the Ouster 64 channel LiDAR with a simple web cam for the calibration. I have modified the topics as the devices I used. However, It seems like I cannot isolate the point cloud data on the chessboard using  rqt_reconfigure_para, even though I changed the config setting from rqt panel, the visualization of point cloud data didn't change. Please see the figure below.
[image]
From the rqt window, the topic that related to point cloud listed on the left is the "feature_extraction". However, from my visualization window, when I change the "Cheesboard" topic to "feature_extraction/cheesboard", there is no point cloud data shown on the right. Only when I select "ouster/point", it works. My understanding is the "feature_extraction" topic tried to convey point cloud data from the LiDAR topic ("ouster/point"), however, it didn't work from my site. So when I use rqt to change the region, the points cannot be isolated. I am not sure if my understanding is correct.
[image]
What should I do to adjust the ROI of the point cloud data?
Thank you

chinitaberrio, 2023-10-08:
@mumu-o 
Here are a few things to check first:
1) Read the readme carefully, showing what needs to be adjusted step by step. Could you ensure you have changed all the topics and extra parameters in the config file?
2)The only topic you must change in rviz is the camera_image.
3)Are you publishing camera_image, camera_info, and point cloud in the same reference time? This package works if the topics are in sync. 
4) Have you calibrated your camera? Are the parameters correct in the camera_info topic? (without the correct intrinsic parameters, the package won't work.)

mumu-o, 2023-10-09:
Hi @chinitaberrio,
Thank you for the reply.
1.1), yes, I have change the topics and modified the camera information based on the result of camera calibration.
[image]
2),yes, I have changed the frame to the ouster_lidar frame, and the camera_image topic to /usb_cam/image_raw
3), I think this is the reason. I recorded a bag file using both lidar and camera topics and the figure below shows the timestamp.
[image]
[image]
2. Yes, I have finished calibrated the camera and changed the parameters based on the camera calibration result. I think the intrinsic parameters should be correct.
I think the timestamp synchronization should be the reason, but I am so confused about how to synchronize the timestamp of two devices. Could you please give me some help?
Thank  you

chinitaberrio, 2023-10-09:
@mumu-o I can't see the camera info topic in your rosbags. Intrinsic calibration is required for the calibration.

mumu-o, 2023-10-09:
@chinitaberrio 
1. the intrinsic parameters are as below.
[image] I have modified the default value with these intrinsic values.
2. I am confused about how to check the header timestamp. If it works if the fps of two devices don't match?
Thank you

chinitaberrio, 2023-10-09:
2) In rqt_bag, select the topic with a right-click and go to view -> raw. Use the arrows in the subsequent panel to navigate. Access the 'header' dropdown, followed by the 'stamp' dropdown to view the timestamp, presented in both seconds and nanoseconds. Perform this for every topic and compare the seconds across all the ROS topics. If the difference is in the magnitude of seconds you need to synchronise your hardware, e.g. for ouster, you can use PTP.

mumu-o, 2023-10-09:
@chinitaberrio Thank you so much, I think this is the reason. The timestamps of my two devices are as follow.
[image] I will work on synchronizing the timestamp.

mumu-o, 2023-10-12:
Hi there,
I'd like to share my solution to a potential issue that Ouster LiDAR users might encounter, as I faced a similar problem. In my setup, I used a simple webcam and the Ouster LiDAR O2.
1. First, ensure you've launched the LiDAR, camera, and calibration topics.
2. Change the frame (in my case, it was "os_sensor"), and the image topic (usb_cam/image_raw in my case). Note that the point cloud should be shown at this time. However, I cannot see the point cloud data.
3. If you're still encountering problems, it may be caused by non-synchronized timestamps from different devices. In my setup, I observed timestamps as shown below:
[image]
4. To address timestamp synchronization issues, follow the guide [PTP](https://static.ouster.dev/sensor-docs/image_route1/image_route2/appendix/ptp-quickstart.html). When configuring the PTP grand master clock, ensure your Ethernet interface supports hardware or software only. If it's software-only, don't forget to change the "time_stamp" from "hardware" to "software". After completing the PTP setup, your timestamps should be synchronized, as shown below:
[image]
5. Restart the LiDAR, camera, and calibration topics, and the package should work correctly:
[image]
Hope this should resolve your issue. Correct me if I am wrong.