acfr/cam_lidar_calibration #6: Error while caliberating os1 64

islamtalha01, 2021-09-30:
Now i am able to extract the checkerboard but i am getting an error due to which i am not able to calibrate. I have attached the error screenshot. Kindly reply as soon as possible.
[image]
[image]

darrenjkt, 2021-09-30:
The issue here is that RANSAC is not able to find your chessboard. In the readme section 2.3.2 "Isolate the chessboard" I showed examples where the chessboard has to be the only visible object in the point cloud. The reason being that this tool has no "smart" identification of where the board is. You have to manually isolate the board. RANSAC will then take the left-most and right-most values of the entire isolated point cloud to estimate the board edges. If there is a lot of noise like 2.3.2 left image or your screenshot, then it'll give the error you are seeing there. 
[image]

The board also has to be tilted as shown in the picture of 2.3.2 right image where it is tilted to look like a diamond. This is so that all the edges of the board can be captured. See Figure 2 in this [paper](https://www.semanticscholar.org/paper/Calibration-between-Color-Camera-and-3D-LIDAR-with-Park-Yun/4479f57a82af8fa2ff502a49a5e9b56edd7fcf85) to understand why we need to tilt the board. We recommend using some sort of mounting solution to suspend the board in the air at a 45 degree tilt so that the point cloud captures nothing else but the board.

islamtalha01, 2021-09-30:
Let me try these things.

islamtalha01, 2021-09-30:
I am running the two drivers of both lidar and camera. But when I start to launch the lidar cam calibration command. It is giving me an error and feature extraction dies just after launch. I am attaching the screenshot. i am trying to do live calibration.

[image]
[image]
[image]

shreyashpalande, 2021-10-05:
@islamtalha01 Check  [this comment](https://github.com/acfr/cam_lidar_calibration/issues/2#issuecomment-934577598). This might be the same reason for you.

islamtalha01, 2021-10-05:
Thanks, I'll check it out.

jogyul00, 2022-11-01:
Can I ask how did you extract the checkerboard from ouster lidar pointcloud?
I'm using ouster 64ch lidar too, but It's point cloud dose not appear on rviz.
 I already modified the ring field.
Help me please