acfr/cam_lidar_calibration #16: Issue with calibration with my own data

sou159, 2022-03-16:
Hi,
I read your paper "Optimising the selection of samples for robust lidar camera calibration" and found it very interesting. I decided to test the available source code. Calibration with your data worked well. However when I tried calibration with my own data using a robosense RS-LiDAR-16 and an ids UI-5280xCP-M mono camera, I faced an issue. In fact, when it comes to feature extraction, The following error appears :
**Connection to node=/feature_extraction failed: param client failed** (Please find below a screen shot of the problem?)

I'm trying to understand the cause of this error, but I don't seem to find it. 
Note1: I can't find "libfeature_extraction.so" in the Lib folder 
Note2: I didn't use a recorded bag file but I used live data instead
Can you please enlighten me on how to solve this problem ?

I would appreciate any help,
Regards,
[image]

arjunskumar, 2022-03-24:
@sou159 are you able to solve?

sou159, 2022-03-29:
@arjunskumar No, I still have a problem.
I tried data acquisition with the docker environment, but an error appears when it comes to features extraction.
[image]

mendoza-G, 2022-04-19:
For note1: I can't find "libfeature_extraction.so" in the Lib folder, it seems the lib is not built successfully, double check the catkin_make log, after deleted build&devel folder. This might be a opencvlib problem.

talhasu, 2022-04-26:
@sou159 Were you able to solve the issue? I am also having the same problem.

duduzai2019, 2023-04-17:
> 注意 1：我在 Lib 文件夹中找不到“libfeature_extraction.so”，似乎 lib 没有构建成功，在删除 build&devel 文件夹后仔细检查 catkin_make 日志。这可能是一个 opencvlib 问题。

can it support opencv3?

closed 2023-06-07 by jclinton830