acfr/cam_lidar_calibration #17: RANSAC PCL erro

mendoza-G, 2022-04-18:
[image]
[image]

This is the erro happened, it seemed to be the issue of setting threshold. And I found that the value in this project was set to 0.004.  Should I change the value? Thanks!

mendoza-G, 2022-04-18:
I' m using robosence M1 lidar, the points are more intensive. Does the brute-force approach take more time? I stucked at "capturing the sample" after clicked the CAPTURE button.

darrenjkt, 2022-07-03:
We tried calibrating a Robosense RS-LIDAR-32 using this tool but we found that it was particularly hard to fit the board using RANSAC due to large distance errors by the lidar. E.g. the distance on black and white vary quite a bit, making the board not very planar in 3D. 

You could adjust the distance threshold as you pointed out but it can lead to a less accurate calibration. However, I guess you could use that as a starting point to tweak the parameters.

closed 2022-07-05 by darrenjkt