acfr/cam_lidar_calibration #18: issue with rosbag play

talhasu, 2022-04-25:
Everything works fine until I run rosbag play -l --pause mybag.bag. I am getting the following error. Could you please take a look and reply me back if you know a way around

[image]

gara-9527, 2022-05-12:
This problem may caused by the `camera_info` topic.
Check if your camera_info topic do exist the intrinsic parameter of your camera.
using `rostopic echo camera_info_topic` to see if this topic contains parameter rather than zero.

aristow1, 2022-10-25:
@gara-9527 im having the same issue, and i checked rostopic echo camera_info_topic and i get parameters. but i still have the same issue

aristow1, 2022-10-25:
@talhasu did you get this solved ?

gara-9527, 2022-10-26:
> @gara-9527 im having the same issue, and i checked rostopic echo camera_info_topic and i get parameters. but i still have the same issue

@aristow1 Can you give me a screenshot of your parameter? Check the parameter and make sure it is correct.For me ,the camera_info_topic's image size doesn't match the real image size cause the issue. 
And this algorithm support online calibration, I personally recommend that you can calibrate online, which can not only save time on recording data, but also facilitate problem solving.

aristow1, 2022-10-26:
@gara-9527 
![cam_param](https://user-images.githubusercontent.com/84501793/197949694-00d4e43d-2eec-4a57-81a9-49ef795a44f4.pn

the parameters are actually 0 here. should i calibrate the camera myself ? with the ros calibration package ?

gara-9527, 2022-10-26:
> @gara-9527 [image]
> 
> the parameters are actually 0 here. should i calibrate the camera myself ? with the ros calibration package ?

Yes,if you don't need high accuracy.I think ros calibration packages is enough.

duduzai2019, 2023-04-17:
> 一切正常，直到我运行 rosbag play -l --pause mybag.bag。我收到以下错误。如果您知道解决方法，能否请您看一下并回复我
> 
> [image]

can you slove it ?

duduzai2019, 2023-04-17:
> 一切正常，直到我运行 rosbag play -l --pause mybag.bag。我收到以下错误。如果您知道解决方法，能否请您看一下并回复我
> 
> [image]

how can you deal it ? i meet the same problem

chinitaberrio, 2023-06-28:
Hi @duduzai2019, Did you solve this problem?

closed 2023-07-13 by chinitaberrio