acfr/cam_lidar_calibration #22: Several Approaches and could not get results

kimjeff-korea, 2022-08-17:
Thanks for your great result and appreciate for your work again. 

I've tried several approaches, tried in ros Melodic for local environment, also on Docker. Finally I tried to make own environment as this blog says. This blog says that there is something to be modified that I tried the same but I could not get the result.

https://programming.vip/docs/cam-for-joint-calibration-of-lidar-and-camera_-lidar_-calibration.html

But the same result. No data shown on the rviz. (I modified the frame id also.) These are what I have modified and progressed.

I am using Ouster OS-1 32 channel. 

**1. 'Ring Match Problem**
I found out that the ring data is not the same as Velodyne and modified the data type as Ouster.

Modified point.h in ouster_ros official package onto uint16_t

[image]

>> Ring not match warning not visible anymore.

**2. Topic Name Configuration**
I've used usb_cam package to publish the camera data on ROS. 
After that I used image_proc as following command.

ROS_NAMESPACE=usb_cam rosrun image_proc image_proc

[image]

RQT Graph shows proper data flow for the extraction

**3. Data not shown on rviz**

Two window appears and expected to adjust the thresholds to show only chessboard. But I could not get anymore of progress.

Can anyone give help for the result? Thanks in advance.

KyunaCho, 2022-09-29:
I'm having the ring match problem. And I'm using os1-64 lidar. Do you know how to remapping the point type?? Please help me.

Eeaeau, 2022-09-29:
I'm also trying to use an ouster lidar and are also stuck at no data showing in RVIZ. As for the datatype, I think he is believe he is referring to the ouster example [ros driver ](https://github.com/ouster-lidar/ouster_example/blob/3a61f8d9c1a804f8d754c49ead8a4a952b77abdf/ouster_ros/include/ouster_ros/point.h#L25) (I archived similar result by using the ouster point inside this repo instead).

jogyul00, 2022-11-25:
Finally I can extract the chessboard with 64-ch ouster lidar. In my case, the ROS LiDAR driver's time was different with ROS time. So I just changed the lidar driver code which is ros.cpp in ''ouster_example'' like the screenshot below.
[image]

closed 2022-12-01 by darrenjkt

kimjeff-korea, 2023-03-13:
In my case, it was something with publishing parameters for camera calibration. When the camera calibration publisher does work well, the codes does a great job. Thanks.