acfr/cam_lidar_calibration #24: The point cloud is not displayed during the calibration of the ouster radar

wangdongxu12345, 2022-10-14:
When I use ouster0-64 lidar for calibration, even if the frame_ The ID and radar topics have been modified, and the data type of ring has been modified to uint16_ t. However, the point cloud cannot be displayed in the calibration interface

jclinton830, 2022-10-19:
@wangdongxu12345 Can you please state which ROS1 version and Ubuntu version you are using?

PS sorry accidentally closed this issue.

closed 2022-10-19 by jclinton830

jogyul00, 2022-11-01:
same issue happens to me.

242726Yuncheng, 2024-06-16:
Same issue happens to me. Have you solved this problem?

jclinton830, 2024-06-18:
> Same issue happens to me. Have you solved this problem?

Can you please state which ROS1 version and Ubuntu version you are using?

242726Yuncheng, 2024-06-20:
melodic and Ubuntu 18.04 LTS