acfr/cam_lidar_calibration #25: error while trying to build the Worspace opencv/cv.hpp

aristow1, 2022-10-17:
Hi everyone,
when i try building my workspace i get this error:
fatal error: opencv/cv.hpp: No such file or directory
   10 | #include <opencv/cv.hpp>
      |          ^~~~~~~~~~~~~~~

im using Ubuntu 20.04 and Ros noetic.
any idea on how i can solve this Problem please ?

aristow1, 2022-10-19:
appearently the solution is to use Ubuntu 18.04 and ROS melodic. with ROS noetic there are some libraries Issue with the code that are outdated

closed 2022-10-19 by aristow1

jclinton830, 2022-10-19:
> appearently the solution is to use Ubuntu 18.04 and ROS melodic. with ROS noetic there are some libraries Issue with the code that are outdated

@aristow1 Sorry about the delay. Yes, this package currently only works for ROS melodic in ubuntu 18.04. Early next year you can expect it to be released for ROS2.

shinianzhihou, 2025-02-28:
dear man, is it released for ros2, which is very important for me~