acfr/cam_lidar_calibration #26: cant visualize the rosbag data for calibration on rviz !

aristow1, 2022-10-25:
so followed the video tutorial you posted on Youtube and followed all the instructions to the word. however when i launch the run_optimiser launch file as stated and then play the Rosbag file i cant visualise any of the data. i changed the topic names in the parameters.yaml file and im even using the same chessboard you used for your calibrations. i get this error and i dont know what the reason is. NEED Help please.
PARAMETERS
 * /camera_info: /camera_info
 * /camera_topic: /image_raw
 * /chessboard/board_dimension/height: 850
 * /chessboard/board_dimension/width: 610
 * /chessboard/pattern_size/height: 7
 * /chessboard/pattern_size/width: 5
 * /chessboard/square_length: 95
 * /chessboard/translation_error/x: 2
 * /chessboard/translation_error/y: 0
 * /feature_extraction/distance_offset_mm: 0
 * /feature_extraction/import_path: /home/****/catki...
 * /feature_extraction/import_samples: False
 * /feature_extraction/num_lowestvoq: 50
 * /feature_extraction/x_max: 10.0
 * /feature_extraction/x_min: -10.0
 * /feature_extraction/y_max: 8.0
 * /feature_extraction/y_min: -8.0
 * /feature_extraction/z_max: 5.0
 * /feature_extraction/z_min: -5.0
 * /lidar_topic: /velodyne_points
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    feature_extraction (cam_lidar_calibration/feature_extraction_node)
    rqt_reconfigure (rqt_reconfigure/rqt_reconfigure)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [21807]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e90796f4-546d-11ed-a450-d04cc1051f2c
process[rosout-1]: started with pid [21833]
started core service [/rosout]
process[feature_extraction-2]: started with pid [21840]
process[rviz-3]: started with pid [21841]
process[rqt_reconfigure-4]: started with pid [21842]
[ INFO] [1666706657.048032589]: Input parameters loaded
[ INFO] [1666706657.094680219]: Reconfigure Request: -10.000000 10.000000 -8.000000 8.000000 -5.000000 5.000000
[ INFO] [1666706657.141405139]: Finished init cam_lidar_calibration
[ WARN] [1666706657.855589389]: MessageFilter [target=velodyne_front_link ]: Discarding message from [unknown] due to empty frame_id.  This message will only print once.
[feature_extraction-2] process has died [pid 21840, exit code -11, cmd /home/*****/catkin_ws/devel/lib/cam_lidar_calibration/feature_extraction_node __name:=feature_extraction __log:=/home/****/.ros/log/e90796f4-546d-11ed-a450-d04cc1051f2c/feature_extraction-2.log].
log file: /home/****/.ros/log/e90796f4-546d-11ed-a450-d04cc1051f2c/feature_extraction-2*.log

jpk219TW, 2022-12-07:
This is a bit of a late reply so I'm not sure if this issue has been solved yet. 
In serviceCB of feature_extractor.cpp, there is a while loop which runs infinitely (not sure what function it serves or why it was included in the first place..). Remove the entire while loop and the package will most likely run fine.

aristow1, 2023-01-09:
@jpk219TW this solved it thank you.

closed 2023-01-09 by aristow1