acfr/cam_lidar_calibration #27: Calibrate LiDAR and camera using files.

peter0617ku, 2022-11-23:
Hello!

I am a beginner in ROS.
There is a question that I want to ask for help.
Through the README tutorial, we know how to use this tool to calibrate our own sensor by playing back **rosbag**.
If I have a lot of image and point cloud files, which are .jpg and .pcd files, can I use these files to calibrate the LiDAR and camera using this GitHub tool?

Thanks for reading patiently!

I'll keep working on that. I will update if there is any progress. Thank you!

aristow1, 2022-12-01:
i dont think you could use .jpg and .pcd files with this repository. you could either use rosbags or live data as well. i think the software needs to know the timestamp of every sample that has been taken and that is known through the header file of each publishing topic. i might be wrong but this is what i think.

darrenjkt, 2022-12-01:
@aristow1 is right. This repository is intended to work with rosbags or live data. It subscribes to the rosmsgs of camera and lidar so the timestamps would have to be about the same. If the timestamps are quite different, the ros ApproximateTimeSync would not be able to find a matching pair of lidar and camera.

closed 2023-06-07 by jclinton830