acfr/cam_lidar_calibration #30: What is the sign of "translation_error"?

LarryDong, 2023-02-20:
Hi Darren, 

Thanks for this codes. It works well for me just after few modifications.

I'm confused by the `translation_error` in configure file. 
It seems to be the "offset" from chessboard center to backing board center. But what is the "positive" direction of axis X and Y? 
I mean, how should I determine the "x" is **"10mm" or "-10mm"** as demonstrated in [2.2-3](https://github.com/acfr/cam_lidar_calibration#22-configuration-files), since the chessboard is symmetric. I'm afraid the detection maybe ambiguous if the chessboard is upside-down when image detecting the coordinates.

jclinton830, 2023-06-07:
Given that your board is symmetric upside down or not should not matter. In either configuration, the software will consider the positive x axis to the right and positive y axis up. So in our case this error is +10 in x and +30 in y axis because the centre of the chessboard is to the top, right of the backing board centre (this is the reference) 

If your chessboard centre is to the left of the backing board centre then your translation_error in x will be negative.

I hope that helps?

LarryDong, 2023-06-07:
Yes, thanks for your reply.

closed 2023-06-07 by LarryDong