acfr/cam_lidar_calibration #31: capture error

zhz03, 2023-02-23:
When we press the "Capture sampe" button, it shows the following errors:
We already modify the checkerboard dimension size to the right one but it still didn't work. Do you know why ?

```
[ INFO] [1677116162.028044308]: Ready for capture

[ INFO] [1677116179.599672785]: Capturing sample
[ INFO] [1677116179.675028523]: Processing sample
[ INFO] [1677116179.701179301]: Chessboard found
[ INFO] [1677116179.702963483]: Publishing chessboard image
[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to optimize model coefficients (0)! Returning the same coefficients.
[pcl::SampleConsensusModelLine::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[pcl::SampleConsensusModelLine::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[pcl::SampleConsensusModelLine::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[pcl::SampleConsensusModelLine::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.

--- Sample 1 ---
Measured board has: dimensions = 880x1200 mm; area = 1.05600 m^2
Distance =  0.00 m
Board angles     =   nan,  nan,  nan,  nan degrees
Board area       = 0.00000 m^2 (-1.05600 m^2)
Board avg height =   0.00mm (-1200.00mm)
Board avg width  =   0.00mm (-880.00mm)
Board dim        =   0.00,  0.00,  0.00,  0.00 mm
Board dim error  = 4160.00

[ERROR] [1677116183.045184085]: Plane fitting error, LiDAR board dimensions incorrect; discarding sample - try capturing again

```

LarryDong, 2023-03-07:
Hi bro,
It seems that you didn't extract the correct chessboard point cloud.
You shoud adjust the threshold to make sure only the chessboard's pointcloud appear in rviz, with as little outliers as possible.\
Hope it helps.

duduzai2019, 2023-04-14:
When we press the "Capture sampe" button, it shows the following errors:
Do you know why ?
[image]
[feature_extraction-1] process has died [pid 93910, exit code -6, cmd /home/duduzai/SPbot/itsc_calib_ws/devel/lib/cam_lidar_calibration/feature_extraction_node __name:=feature_extraction __log:=/home/duduzai/.ros/log/51abcb48-da75-11ed-8ebb-97b7bce6879a/feature_extraction-1.log].
log file: /home/duduzai/.ros/log/51abcb48-da75-11ed-8ebb-97b7bce6879a/feature_extraction-1*.log
^C[rqt_reconfigure-3] killing on exit

jclinton830, 2023-06-07:
> > ou shoud adjust the threshold to make sure only the chessboard's pointcloud appear in rviz, with as little outliers as possible.
> > Hope it helps.
> 
> I adjusted already but the problem is still there.

Can you please post a picture of your pointcloud after adjusting the sliders to filter everything else other than the board?

j-saez, 2023-06-16:
Hello,

I am experiencing a similar problem. I am using ROS noetic on Ubuntu 20.04 to launch the publisher nodes. To run you package I use the docker image you provide. The lidar I am using is a velodyne VLP-32C and the camera is a stereo camera ZED 2.

The  initial point cloud is: 
[image]

After filtering it (Do not pay attention to the lidar warning, as they were given before running rviz):
[image]

I save a rosbag with the data to be able to reproduce it. It can be found [here](https://studentmailuwsac-my.sharepoint.com/:u:/g/personal/javier_saez-perez_uws_ac_uk/EY0mz9QDtgNPquF9DI9TStsBiIGw4T2p86BTOwCY_d1BFg?e=70dZqy).

Also, the content of cfg/params.yaml is:

```bash
# Topics
camera_topic: "/zed2/zed_node/rgb_raw/image_raw_color"
camera_info: "/zed2/zed_node/rgb_raw/camera_info"
lidar_topic: "/velodyne_points"

# Dynamic rqt_reconfigure default bounds
feature_extraction:
  x_min: -10.0
  x_max: 10.0
  y_min: -8.0
  y_max: 8.0
  z_min: -5.0
  z_max: 5.0

# Properties of chessboard calibration target
chessboard:
  pattern_size:
    height: 11
    width: 13
  square_length: 87
  board_dimension:
    width: 1222
    height: 1045
  translation_error:
    x: 0
    y: 0

```

chinitaberrio, 2023-06-18:
Hi @j-saez. Try rotating the chessboard such that it is in the shape of a diamond (at an angle of 45° with respect to the ground)

j-saez, 2023-06-20:
I have tried but it did not work either:
[image]

Any more suggestions? I do not know what I am doing wrong, as I have seen in the README.md that the authors were able to calibrate a VLP-16.

chinitaberrio, 2023-06-20:
Have you ensured that the configuration files have been thoroughly reviewed and accurately match the dimensions of your board?
https://github.com/acfr/cam_lidar_calibration#22-configuration-files

closed 2023-08-09 by jclinton830

youzhihh, 2023-10-18:
> When we press the "Capture sampe" button, it shows the following errors: Do you know why ? [image] [feature_extraction-1] process has died [pid 93910, exit code -6, cmd [...]

If this happens when you run the data set that comes with the project (roslaunch cam_lidar_calibration run_optimiser.launch import_samples:=true), it is a problem with the opencv version. If you can run the data set that comes with the project, but there is a problem with your own data set, it will be very serious. It may be that the topic of camera_info you set in the parameters is incorrect, or the content of the topic is incorrect (the contents are all 0, or the topic is not published at all). You can write a node to publish camera_info (camera internal parameters and other parameters).