acfr/cam_lidar_calibration #42: The calibration board has been filtered out. But the RANSAC unsuccessful occurs if i click capture sample

gzchenjiajun, 2023-08-10:
One is before the filter and the other is after the filter, which has been placed at 45°
[image]
[image]

But RANSAC errors still occur, why?

[ INFO] [1691641310.418340591]: Ready for capture
 
[ INFO] [1691641333.268849203]: Capturing sample
[ INFO] [1691641333.414355339]: Processing sample
[ INFO] [1691641333.422643699]: Chessboard found
[ INFO] [1691641333.423457583]: Publishing chessboard image
[ WARN] [1691641333.851872087]: Still working with last sample!
[pcl::SampleConsensusModel::getSamples] Can not select 0 unique points out of 1!
[pcl::RandomSampleConsensus::computeModel] No samples could be selected!
[pcl::SACSegmentation::segment] Error segmenting the model! No solution found.
[pcl::SampleConsensusModel::getSamples] Can not select 0 unique points out of 1!
[pcl::RandomSampleConsensus::computeModel] No samples could be selected!
[pcl::SACSegmentation::segment] Error segmenting the model! No solution found.
[ERROR] [1691641334.313363673]: RANSAC unsuccessful, discarding sample - Need more lidar points on board

The following is my yaml information

 Topics
camera_topic: "/bonson/camera_image"
camera_info: "/bonson/camera_info"
lidar_topic: "/bonson/point_cloud"

 Dynamic rqt_reconfigure default bounds
feature_extraction:
  x_min: -10.0
  x_max: 10.0
  y_min: -10.0
  y_max: 10.0
  z_min: -10.0
  z_max: 10.0
  
 Properties of chessboard calibration target
chessboard:
  pattern_size:
    height: 11
    width: 8  
  square_length: 35
  board_dimension:
    width: 500
    height: 380
  translation_error:
    x: 2
    y: 0

——————
distortion_model: "fisheye"
width: 1920
height: 1080
D: [-0.958013458, -28.3233961, 0.0736144326, -0.0146417112, 505.177751]
K: [1012.71567, 0.0, 316.010119, 0.0, 1047.60412, 178.325968, 0.0, 0.0, 1.0]

gzchenjiajun, 2023-08-16:
[image]
still  RANSAC unsuccessful

gzchenjiajun, 2023-08-16:
Is the data from my lidar  not compatible with this calibration repo?

chinitaberrio, 2023-10-08:
@gzchenjiajun The problem you're having is due to your setup, you're holding the checkerboard thus some points of your hands and arms might be included. In our experiments we use a stand to place the board so only the board is detected by the lidar.

GodLiu1, 2023-11-06:
May I ask how you filtered the point cloud? I adjusted rqt_reconfigure, but the point cloud did not change

chinitaberrio, 2023-11-08:
@GodLiu1, please refer to [this](https://github.com/acfr/cam_lidar_calibration/issues/4#issuecomment-1758757785) thread.

jclinton830, 2023-12-12:
@GodLiu1 I am closing this due to inactivity. Feel free to open it again if you still have issues.

closed 2023-12-12 by jclinton830