acfr/cam_lidar_calibration #44: Reprojection Image Seems Strange

monamourvert, 2023-08-16:
Hi, I applied your code to my own data an strangely I got this result. I worked with your noetic branch with Azure Kinect Camera and Ouster Lidar.

My reprojection result:
[image]

Since I'm new at this, I just wondering is this kind of result is normal? I tried several times, and most of them produce similar result. The only different is the place of the yellow and the green cross sometimes not lies on the center of board. Is there anyone also experience this strange result?

monamourvert, 2023-08-16:
Another thing that I found, no matter perfectly I isolate the chessboard, the dimension error is quite large. 
So, to summarize it:
1. I calculate the offset of the chessboard as accurate as I can for the params.yaml
2. The camera_info is directly retrieved from /rgb/camera_info from Azure Kinect so I assume there is nothing wrong with the intrinsic parameter.
3. I tried to isolate the chessboard as detail as I can. This one is just an example (I screenshot this as fast as I can)
[image]
[image]
But I guess the dot that mark the board area seems having a big translation from the edge. I don't I'm not so sure.
4. The dimension error of the board is strangely always > 40
[image]

Hope somebody can help me about this. Thanks!

chinitaberrio, 2023-10-08:
@monamourvert, which lidar are you using? is the camera calibrated?

closed 2023-10-18 by chinitaberrio