acfr/cam_lidar_calibration #48: Error when running the quick start

juannn01, 2023-11-16:
Hi! I am facing an error when running the calibration process using the quick-start data provided. A snapshot of the error is shown below. Look forward for your reply! Thanks in advance! (Additional information: ROS Noetic, Ubuntu 20.04) 
[image]

YHuniv, 2023-12-11:
This resolved my problem, because i am using opencv 4.2.0 :
change #include <opencv/cv.hpp> to  **#include <opencv2/opencv.hpp>**
in **optimiser.h.**
(/catkin_ws/src/cam_lidar_calibration/include/cam_lidar_calibration/optimiser.h)
and change  **optimiser.cpp:**  
from 
// cv::reduce(trans_diff, summed_diff, 1, CV_REDUCE_SUM, CV_64F);

// to 
**cv::reduce(trans_diff, summed_diff, 1, cv::REDUCE_SUM, CV_64F);**

(/catkin_ws/src/cam_lidar_calibration/src/optimiser.cpp)

jclinton830, 2023-12-12:
@YHuniv thanks for your input to the issue. 

If that does not work, please feel free to reopen this issue. I am closing it as done for now.

closed 2023-12-12 by jclinton830

natik-bw, 2024-04-20:
Reopening this topic, as the solution didn't resolve my error.

- I am using ros-noetic, with Opencv-4.5.5 (due to other project dependencies)
- Changing `#include <opencv/cv.hpp>' to '#include <opencv2/opencv.hpp>
in optimiser.h.` didn't resolve the problem, since it caused the following error when executing `catkin build`:
[image]

I was able to successfully build when I replaced `#include <opencv/cv.hpp>' to:
```
#include <opencv2/core/core.hpp>
#include <opencv2/imgcodecs.hpp>
#include <opencv2/calib3d.hpp>
```

But then, still got the same runtime error when executing the quick start
```roslaunch cam_lidar_calibration run_optimiser.launch import_samples:=true```

[image]

*** logs ***
[rosout-1-stdout.log](https://github.com/acfr/cam_lidar_calibration/files/15047083/rosout-1-stdout.log)
[rosout.log](https://github.com/acfr/cam_lidar_calibration/files/15047084/rosout.log)
[roslaunch-netanel-Precision-5560-26735.log](https://github.com/acfr/cam_lidar_calibration/files/15047085/roslaunch-netanel-Precision-5560-26735.log)
[master.log](https://github.com/acfr/cam_lidar_calibration/files/15047086/master.log)

***NOTE*** 
I don't know if this is correct, but I used the `calib_v2` branch, and there 
cv::reduce(...) under optimiser.cpp [here](https://github.com/acfr/cam_lidar_calibration/blob/calib-v2/src/optimiser.cpp#L456)
was not commented out (all I had to do was replace CV_REDUCE_SUM by cv::REDUCE_SUM for proper compatibility), so I wasn't sure whether I missed an additional change I had to do.
It wasn't clear to me from @YHuniv 's instructions.