acfr/cam_lidar_calibration #50: I am getting pcl::PCLBase<pcl::PointXYZIR> and undefined reference to `pcl::KdTreeFLANN<pcl::PointXYZIR, flann::L2_Simple<float> >::KdTreeFLANN(bool) while library linking

anamadheya, 2024-01-18:
am using ROS Noetic to build some calibration code and I am facing

undefined reference to pcl::PCLBase<pcl::PointXYZIR>::deinitCompute()' undefined reference to pcl::PCLBasepcl::PointXYZIR::initCompute()'
undefined reference to pcl::PCLBase<pcl::PointXYZIR>::setIndices(boost::shared_ptr<std::vector<int, std::allocator<int> > const> const&)' pcl::KdTreeFLANN<pcl::PointXYZIR, flann::L2_Simple<float> >::KdTreeFLANN(bool)' undefined reference to pcl::PCLBasepcl::PointXYZIR::setInputCloud(boost::shared_ptr<pcl::PointCloudpcl::PointXYZIR const> const&)'
/catkin_ws/devel/.private/cam_lidar_calibration/lib/libcam_lidar_calibration.so: undefined reference to pcl::KdTreeFLANN<pcl::PointXYZIR, flann::L2_Simple<float> >::setSortedResults(bool)' catkin_ws/devel/.private/cam_lidar_calibration/lib/libcam_lidar_calibration.so: undefined reference to pcl::PCLBasepcl::PointXYZIR::setIndices(unsigned long, unsigned long, unsigned long, unsigned long)'
catkin_ws/devel/.private/cam_lidar_calibration/lib/libcam_lidar_calibration.so: undefined reference to pcl::PCLBase<pcl::PointXYZIR>::setIndices(boost::shared_ptr<pcl::PointIndices const> const&)' catkin_ws/devel/.private/cam_lidar_calibration/lib/libcam_lidar_calibration.so: undefined reference to pcl::PCLBasepcl::PointXYZIR::PCLBase()'
catkin_ws/devel/.private/cam_lidar_calibration/lib/libcam_lidar_calibration.so: undefined reference to `pcl::PCLBasepcl::PointXYZIR::setIndices(boost::shared_ptr<std::vector<int, std::allocator > > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/feature_extraction_node.dir/build.make:303: catkin_ws/devel/.private/cam_lidar_calibration/lib/cam_lidar_calibration/feature_extraction_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:3731: CMakeFiles/feature_extraction_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

anamadheya, 2024-01-19:
Please refer https://github.com/PointCloudLibrary/pcl/issues/5933 if you get something like this

closed 2024-01-19 by anamadheya

jclinton830, 2024-04-18:
> Please refer [PointCloudLibrary/pcl#5933](https://github.com/PointCloudLibrary/pcl/issues/5933) if you get something like this

Glad you figured it out. Which branch were you testing?