acfr/cam_lidar_calibration #53: frame_id error Causing RViz Freeze During Background Capture

jiheddachraoui, 2024-02-25:
Hello , I'm trying out the offline mode with a saved bag. I followed the instructions in the README and updated the config file as directed. But when I hit the "capture background" button, Rviz freezes and shuts down. I also got this warning:

```
[INFO] [1708884414.136332641]: Capturing background pointcloud
[WARN] [1708884414.137188861]: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty

```
I checked the LiDAR topic, and the frame_id isn't empty.
[image]

tui-abdul, 2024-04-03:
> Hello , I'm trying out the offline mode with a saved bag. I followed the instructions in the README and updated the config file as directed. But when I hit the "capture background" button, Rviz freezes and shuts down. I also got this warning:
> 
> ```
> [INFO] [1708884414.136332641]: Capturing background pointcloud
> [WARN] [1708884414.137188861]: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
> ```
> 
> I checked the LiDAR topic, and the frame_id isn't empty. ![Screenshot from 2024-02-25 18-38-33](https://private-user-images.githubusercontent.com/93733459/307612853-16ae61c8-3723-4749-b0b9-8b1beb7e6618.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIxNTI3NzEsIm5iZiI6MTcxMjE1MjQ3MSwicGF0aCI6Ii85MzczMzQ1OS8zMDc2MTI4NTMtMTZhZTYxYzgtMzcyMy00NzQ5LWIwYjktOGIxYmViN2U2NjE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAzVDEzNTQzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyMjFkZmIyMzQxMDM1ZjA2YTc0NGFjNzAzN2FkODM4MTNkNjlkNDgzOTQ2MDI3M2Y4NWVjYzA1MDNjN2Q4YTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.m5hDhAlorjAbl5UX2LkpITnOwown-GbPve5pfDUR3Fs)

I have the same issue. Did you solve it?

jiheddachraoui, 2024-04-05:
@tui-abdul  hello, i decised to work with docker finally , and it works.

closed 2024-06-18 by jclinton830