acfr/cam_lidar_calibration #60: Strange issue when using my own data.

ShiLong0526, 2024-06-07:
Hi, thanks for your work. I meet some error when I calibrate with my won data. The error occurs when I begin to play the ros bag. The detail is as follow:

> fsl@fsl-All-Series:~/catkin_ws$ roslaunch cam_lidar_calibration run_optimiser.launch import_samples:=false
> ... logging to /home/fsl/.ros/log/c6b51808-249e-11ef-8877-5f14ede982f9/roslaunch-fsl-All-Series-33406.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://fsl-All-Series:40139/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /camera_info: /galaxy_camera/ga...
>  * /camera_topic: /galaxy_camera/ga...
>  * /chessboard/board_dimension/height: 910
>  * /chessboard/board_dimension/width: 460
>  * /chessboard/pattern_size/height: 8
>  * /chessboard/pattern_size/width: 6
>  * /chessboard/square_length: 35
>  * /chessboard/translation_error/x: 5
>  * /chessboard/translation_error/y: 5
>  * /feature_extraction/distance_offset_mm: 0
>  * /feature_extraction/import_path: /home/fsl/catkin_...
>  * /feature_extraction/import_samples: False
>  * /feature_extraction/num_lowestvoq: 50
>  * /feature_extraction/x_max: 10.0
>  * /feature_extraction/x_min: -10.0
>  * /feature_extraction/y_max: 8.0
>  * /feature_extraction/y_min: -8.0
>  * /feature_extraction/z_max: 5.0
>  * /feature_extraction/z_min: -5.0
>  * /lidar_topic: /rslidar_points
>  * /rosdistro: noetic
>  * /rosversion: 1.16.0
> 
> NODES
>   /
>     feature_extraction (cam_lidar_calibration/feature_extraction_node)
>     rqt_reconfigure (rqt_reconfigure/rqt_reconfigure)
>     rviz (rviz/rviz)
> 
> auto-starting new master
> process[master]: started with pid [33433]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to c6b51808-249e-11ef-8877-5f14ede982f9
> process[rosout-1]: started with pid [33462]
> started core service [/rosout]
> process[feature_extraction-2]: started with pid [33469]
> process[rviz-3]: started with pid [33470]
> process[rqt_reconfigure-4]: started with pid [33471]
> [ INFO] [1717744983.691498308]: Input parameters loaded
> [ INFO] [1717744983.721899562]: Reconfigure Request: -10.000000 10.000000 -8.000000 8.000000 -5.000000 5.000000
> [ INFO] [1717744983.767626615]: Finished init cam_lidar_calibration
> _**[feature_extraction-2] process has died [pid 33469, exit code -11, cmd /home/fsl/catkin_ws/devel/lib/cam_lidar_calibration/feature_extraction_node __name:=feature_extraction __log:=/home/fsl/.ros/log/c6b51808-249e-11ef-8877-5f14ede982f9/feature_extraction-2.log].**_
> log file: /home/fsl/.ros/log/c6b51808-249e-11ef-8877-5f14ede982f9/feature_extraction-2*.log
> Exception in thread Thread-8:
> Traceback (most recent call last):
>   File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 567, in connect
>     self.socket.connect((dest_addr, dest_port))
> ConnectionRefusedError: [Errno 111] Connection Refused
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 509, in call
>     transport.connect(dest_addr, dest_port, service_uri)
>   File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 596, in connect
>     raise TransportInitError(str(e)) #re-raise i/o error
> rospy.exceptions.TransportInitError: [Errno 111] Connection Refused
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/opt/ros/noetic/lib/python3/dist-packages/dynamic_reconfigure/client.py", line 222, in update_configuration
>     msg = self._set_service(config).config
>   File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
>     return self.call(*args, **kwds)
>   File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 512, in call
>     raise ServiceException("unable to connect to service: %s"%e)
> rospy.service.ServiceException: unable to connect to service: [Errno 111] Connection Refused
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
>     self.run()
>   File "/opt/ros/noetic/lib/python3/dist-packages/rqt_reconfigure/param_updater.py", line 95, in run
>     raise exc
>   File "/opt/ros/noetic/lib/python3/dist-packages/rqt_reconfigure/param_updater.py", line 90, in run
>     self._reconf.update_configuration(configs_tobe_updated)
>   File "/opt/ros/noetic/lib/python3/dist-packages/dynamic_reconfigure/client.py", line 224, in update_configuration
>     raise DynamicReconfigureCallbackException('service call failed')
> dynamic_reconfigure.DynamicReconfigureCallbackException: service call failed

I tried to calibrate on virtual computer and real computer, but all of them have this issue. 
Thank you!

ShiLong0526, 2024-06-07:
[image]
[image]

jclinton830, 2024-06-18:
@ShiLong0526 Can you please check which branch of the repo are you currently using? Since you are on ROS Noetic, you need to checkout noetic_v2, if you have already got noetic_v2, please make sure to update it by doing a git pull. See if that fixes your issue.

closed 2024-06-18 by jclinton830