acfr/cam_lidar_calibration #63: Can't filter point cloud with rqt

EMIR-Mh01, 2024-07-19:
[image]
i work with velodyne as lidar and an optical camera and i want the topic of features_extraction to appear in rviz but the only points that appear are velodyne_points i want the calibration with my own data .   I have changed the params.yaml and camera_info.yaml according to my dataset.                                                                                                                                                                           distortion_model: "plumb_bob"
width: 1280
height: 720
D: [0.137822,0.35667,-0.035565,0.006467]
K: [975.093708  , 0,      649.525527 ,0.       ,  980.29991878, 310.223634,0.           0.           1.        ]]
# Topics
camera_topic: "/rtsp_stream"
camera_info: "/camera_info"
lidar_topic: "/velodyne_points"

# Dynamic rqt_reconfigure default bounds
feature_extraction:
  x_min: 3.0
  x_max: 7.0
  y_min: -3.0
  y_max: 3.0
  z_min: -1.7
  z_max: 0.0
  k: 50
  z: 1
  voxel_res: 0.4

# Properties of chessboard calibration target
chessboard:
  pattern_size:
    height: 7
    width: 5
  square_length: 70
  board_dimension:
    width: 45.7
    height: 58.7
  translation_error:
    x: 2
    y: 2                                                                                                                                                                                                                         I am currently trying your v2 package as well. Still the dynamic rqt window can't apply to my data. I can't see the pointcloud data change after I apply the filter. Do I suppose to wait longer? Btw, I made to few changes to let the package compile with my ROS Noetic version.

zayedrmdn, 2024-08-14:
Has this issue been solved? @EMIR-Mh01

chengziNaomi, 2025-09-15:
Hello, I'm also encountering the same issue now. In my Rivz, it shows the point cloud of the lidar instead of the  features_extraction. Could you please tell me how you solved it?