acfr/cam_lidar_calibration #67: Can't quick start to run the calibration process

Abelcannotcode, 2024-10-16:
I am currently working with OpenCV version 4.5.5 on Ubuntu 18.04, and the package has been tested with ROS Melodic. When I execute the command roslaunch cam_lidar_calibration run_optimiser.launch import_samples:=true, I encounter the following error:
[feature_extraction-2] process has died [pid 78353, exit code -11, cmd /home/user/cam_lidar_calib/catkin_ws/devel/lib/cam_lidar_calibration/feature_extraction_node __name:=feature_extraction __log:=/home/user/.ros/log/aa9a20da-8b84-11ef-b4ca-000c2901b696/feature_extraction-2.log].
log file: /home/user/.ros/log/aa9a20da-8b84-11ef-b4ca-000c2901b696/feature_extraction-2*.log
[image]
hments/assets/bd38d6a6-76b8-4633-b21e-bc85d7c612f6)
I have attempted to resolve this issue by testing various versions of OpenCV (including 4.10.0 and 3.4.16), but I have been unsuccessful. I appreciate your dedication to this project, as it holds significant value in sensor calibration. However, I am facing challenges and would greatly appreciate your assistance in troubleshooting this issue. I am uncertain whether it is related to my environment.

oswanpy, 2024-12-06:
did you find any solution for the issue

JinXJinLeX, 2025-02-18:
I met the same problem, exactly like yours. I changed the version of OpenCV to 3.4.1( I always use this version) in Cmakelists.txt, maybe you should delete build and devel folder then recompile the project.